rricharz/pi-top-battery-status #9: Issues with gtk_battery after Update : Hardware not matching expected BCM

rfv-370, 2017-02-26:
Hi Rich,
I am getting:
$ gtk_battery
Unable to determine hardware version. I see: Hardware   : BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.

I can confirm that this a genuine Rpi3 (I think) - is there a way to force gtk_battery to start anyway ?

Best,
Robert

rricharz, 2017-02-27:
Hi Robert,

I have no idea what's causing this problem. Wiringpi is used in gtk-battery. Can you please check your version number of wiringpi with
`sudo apt-cache policy wiringpi`

The latest version on Raspbian is 2.32. If you have an earlier version, do a 
`sudo apt-get install wiringpi`

Please let me know the outcome. If you have a genuine Raspberry Pi, you might write to projects@drogon.net and report the outcome here.

Best Regards, Rene

rricharz, 2017-03-04:
Any news on this issue? I am not aware of anybody else observing it.

Best regards, Rene

rfv-370, 2017-03-07:
Thank you. I update/upgraded, reinstalled and then all came back up normal.BestRobert

 Envoyé le : Samedi 4 mars 2017 13h08

closed 2017-03-07 by rricharz