rricharz/pi-top-battery-status #10: Add instructions for registering battery device?

bcnjr5, 2017-03-03:
Hello,

I have recently figured that the `sbs-battery` kernel module (located [here](https://github.com/torvalds/linux/blob/master/drivers/power/supply/sbs-battery.c)) can read the Pi-Top battery.

When registered, the battery is accessible from `/sys/class/power_supply/*`, so users will not require a special tool to access it.
In my case, this allows the `tint2` taskbar to automatically display the battery level in the bottom right.
It stands to reason that other, more common desktop environments will do the same.

The only problem is that the battery sends strange values half the time, so you might get a display alternating between `0%` and the real value in most cases. This seems to be because of a faulty battery though, as I'm one of the people who needed to use the firmware update tool many times. (Many thanks for [battery_problems.md](https://github.com/rricharz/pi-top-battery-status/blob/master/battery_problems.md)!)

As I am not a kernel developer, it took a while to figure out how to register the device with driver.
Could you perhaps add this information to your guide so that other people will have less trouble following a similar route?

One of the main reasons I'm asking you is because it seems that every time I search the internet for information about the Pi-Top battery, I find this repository in the results.

The command to register the battery is as follows (run as root):

````sh
# make sure the sbs_battery module is loaded first!
echo sbs-battery 0x0b > /sys/class/i2c-adapter/i2c-1/new_device
````

As the `sbs_battery` module was not included in my copy of Raspbian, I had to build it myself.
This is what I put in my `/etc/rc.local`:

````sh
# Register Pi-Top battery
# I put the module in /usr/local/share/ptb, but that's not required.
insmod /usr/local/share/ptb/sbs-battery.ko
echo sbs-battery 0x0b > /sys/class/i2c-adapter/i2c-1/new_device
````

bcnjr5, 2017-03-03:
The Pi-Top doesn't seem to have any documentation.
If someone out there knows what is wrong with the batteries, I might be able to write a module that targets the pi-top specifically and doesn't return such wild values.

rricharz, 2017-03-05:
We are currently looking into using a kernel module. The problem is the i2c reliability of the rpi talking to the battery pack. The standard sbs-battery module does not handle this properly.

rricharz, 2017-03-06:
Instructions to install this kernel module and i2c "bitbanging" can be found at https://github.com/bcnjr5/linux-pitop-battery

I have tested this approach and it works nicely together with the standard battery panel widget of the standard Raspbian desktop. But be aware that other i2c programs will not work, including the pi-topSPEAKER and the battery display in pi-topOS.

closed 2017-03-06 by rricharz