rricharz/pi-top-battery-status #17: V1.5 still won't autostart on my setup

BKrobusek, 2017-09-19:
I followed your instructions for "Using the Program as a user other than pi". I also went to the web page you referenced for setting the user privileges appropriately. When I downloaded the latest zip and tried to run install I got the following errors:

grep: /root/.config/lxsession/LXDE-pi/autostart: No such file or directory
cp: cannot stat '/root/.config/lxsession/LXDE-pi/autostart': No such file or directory
./install: 24: ./install: cannot create /root/.config/lxsession/LXDE-pi/autostart: Directory nonexistent

I created the necessary folder structure and install ran without errors, and it immediately ran the app. But logging back in still doesn't automatically start the program. Running install again brings up the app.

I will admit to being a Linux newbie, so there may be something obvious that I've missed.

rricharz, 2017-09-19:
I don't understand your setup, but it appears that you are running the installation program being root. That will not work. I assume that you are starting the desktop as a user with a home directory in /home, not root. Once the desktop is ready, open a terminal on the desktop and type 'whoami'. That will give you the name of the user, who has started the desktop. Furthermore, do not switch to the user root at this time with 'su root' or similar! Rather start the installation script in this terminal on the desktop as the user who has started the desktop. It will then initialize everything properly.

Again, being the user who has started the desktop, not root, you can type the following:
  cd ~/.config/lxsession/LXDE-pi
 cat autostart

and you should see the call to gtk_battery, which is made when the desktop starts as this user.

I hope this helps. If not, please let me know more details about your special setup.

BKrobusek, 2017-09-19:
Well, I don't think I'm logged in as root. When I login and run whoami I see the user name I created:: bkrobusek. When I follow your instructions to cat autostart I see the call to gtk_battery as the last line of the file. If I enter the gtk_battery command from the terminal prompt I get an error: "bash: dev/null: No such file or directory".

rricharz, 2017-09-19:
OK, we are getting somewhere.
Does
  /home/bkrobusek/bin/gtk_battery
work, and
  gtk_battery
fail?

If yes, what does
  $PATH
show? Does it include /home/bkrobusek/bin?

BKrobusek, 2017-09-19:
If I move to /home/bkrobusek/bin and execute gtk_battery I get a segmentation fault error. $PATH shows that /home/bkrobusek/bin is in the path. It's actually the first entry. If I run install again the app is installed.

rricharz, 2017-09-19:
Sorry, I have no further clue what's happening. It works fine on my system in any account. It mever tries to access /root.

BKrobusek, 2017-09-19:
I appreciate your efforts! I'll keep futzing with it and see what I can come up with.

Update: I'm baffled. It just started working. I tried running gtk_battery from Terminal, and saw a couple of locale errors. So I updated the settings in Raspberry Pi Configuration -> Localisation, and rebooted. And now it works. But I still get the locale errors if I try to run gtk_battery from Terminal.

I have no idea... But it seems to be working now. Thanks for your help!

BKrobusek, 2017-11-14:
Rene:

I hesitate to post this on GitHub as a bug because I’m not confident that it is. But I wanted to ask you in an e-mail to see if you had some other ideas.

It seems that, after running your pi-top-setup, that I’m no longer able to save changes to the GUI. I can make changes easily enough, and I do not get any errors when closing the preference changes windows, but the changes never stick on the next reboot.

For instance, if I go to Desktop Preferences and change the font size under the System tab to 10, and click okay, the font and icon sizes change. But after shutting down or rebooting, the icons and font sizes return to their previous values.

This is using a Version 1 Pi-Top that’s running Raspbian GNU/Linux 9 (stretch), kernel version Linux 4.9.59-v7+.

I’m wondering if the permissions for the Linux GUI configuration file may have gotten changed, but I don’t even know what file to look for, or where to look for it, to see if that could be the problem.

Do you have any suggestions?

And FWIW, I would like to thank you for all of the work you’ve put into supporting the Pi-Top product. I’m using it a lot more because of your improvements and additions!

Thanks!

	Bruce

Bruce A. Krobusek
Farmington  NY

rricharz, 2017-11-14:
Hi Bruce,
I have checked your observation and can confirm the problem. I think that this is caused by some software from pi-top which is installed together with their pt-hub software. I will contact them right away to find out what can be done about it.
Rene

BKrobusek, 2017-11-14:
Rene:

Thank you so much for confirming my problem! As I said earlier, you are an amazing resource. Thanks for all that you do!

     Bruce

Bruce A. Krobusek
Farmington  NY

rricharz, 2017-11-14:
Buce,

The problem can be fixed permanently with the following command:
`sudo systemctl disable pt-display`

pi-top will fix this hopefully in the next release of their "driver software". It has to do with their messaging service, which displays low battery warnings on the desktop. These messages will continue to work after applying the fix above.

Let me please know if this does not fix your problem, or if you ave another issue with the software.

Rene

BKrobusek, 2017-11-14:
Rene:

That seems to work. Thanks again!

	Bruce

Bruce A. Krobusek
Farmington  NY